spartanracingelectric/VCU #11: 80 kW limit

chucktesta154k, 2015-11-10:
We need to be able to limit the current, or torque commanded, such that at higher speeds we do not draw more than 80 kW from the batteries.

GuySmily, 2015-12-21:
BMS only provides 1s average current measurement - could get us into trouble with 100ms/500ms rules (EV2.2.4)

chucktesta154k, 2015-12-21:
we just make sure that we are decently well below it, and in testing can
figure out how much we can try to creep up on it, but we can just stay a
bit under it.

Charlie Trevaskis
San Jose State University
Mechanical Engineering Department
Spartan Racing Electric | Chief Engineer
(209) 217-6427
CharlieTrevaskis@gmail.com <charlietrevaskis@gmail.com>
http://www.spartanracingelectric.com

rabeelio, 2016-05-14:
done

closed 2023-09-22 by shibcreate