spartanracingelectric/VCU #31: Add HVIL shutdown relay

GuySmily, 2016-02-06:
Pin 132 to NC relay, open relay to cut power e.g. upon loss of BMS communication

chucktesta154k, 2016-02-06:
should this be NO for safety? or it can be NC because it isn't required

Charlie Trevaskis
San Jose State University
Mechanical Engineering Department
Spartan Racing Electric | Chief Engineer
(209) 217-6427
CharlieTrevaskis@gmail.com <charlietrevaskis@gmail.com>
http://www.spartanracingelectric.com

guill0tine, 2016-02-07:
Yea for safety, we should be going with the NO relay. But rules don't care as its not required in the HVIL loop.

GuySmily, 2016-02-14:
Pin 132 connected to NO relay.
On VCU startup: Set Pin 132 = ??
On HVIL Term Sense == high: Set Pin 132 = ??
----
If ??? (e.g. > #ms since last BMS CAN message), set pin low
If ???, set pin high again?

GuySmily, 2016-02-24:
Duplicate of #9

GuySmily, 2016-02-26:
Pin 132 = IO_DO_01

closed 2016-11-22 by GuySmily