spartanracingelectric/VCU #35: Create Throttle Calibration Function

vinayponnaganti, 2016-02-16:


GuySmily, 2016-02-16:
Throttle and brake calibration are in the same function calibrateTPS() in sensorCalculations.c.

TPS calibration is okay as written, I think.

To do: Decide on brake pressure calibration/curve.  What's 100%?  As hard as the driver can press?  How much braking needed to trigger BSPC or to allow RTD?

Suggestion: Let sensor max value be 100%.  Save calibration max (hardest the driver presses during calibration function).  Each driver should calibrate before driving, but previous calibration should be saved to EEPROM in case of e.g. momentary power loss.  Have easily-changable values for RTD and BSPC pressure requirements.

GuySmily, 2016-02-24:
See also #23.

closed 2023-09-22 by shibcreate