spartanracingelectric/VCU #36: Change Torque Encoder Plausibility Check to disable inverter instead of command 0 torque?

GuySmily, 2016-02-17:
Issue: Currently/temporarily, the TEPC commands 0 torque but does not shut down the inverter.  This will not comply with rules, as the motor controller still applies some power at 0 torque as evidenced by the motor bumping around slightly on the bench.

Todo: Disable the inverter (and don't change the torque command anymore?)
Todo: Figure out restart procedure.  Re-enable automatically once error condition goes away?  Require RTD button to enable inverter?
Todo: Do more testing with re-enabling inverter while motor is still spinning down.  On the bench there is a hard force applied and there may be a risk of damage to the motor/controller.

EV2.3.5 If an implausibility occurs between the values of these two sensors the power to the motor(s) must be immediately shut down completely. It is not necessary to completely deactivate the tractive system, the motor controller(s) shutting down the power to the motor(s) is sufficient.

chucktesta154k, 2016-02-17:
isn't there an inverter disable/enable bit on the can message? I thought the plan of record was to use that.

We can not require the RTD button to be pressed again. That would be ridiculous, and potentially fatal.

GuySmily, 2016-02-22:
Yes, there is an inverter enable/disable bit.  However, if you disable the inverter and re-enable the inverter while the motor is still rotating, it physically shocks the motor pretty hard.  It sounds like something is slamming into the workbench.  I'll setup demo code for the next meeting if I have enough time.

chucktesta154k, 2016-02-22:
Ohhh. Okay. Then I'm not sure about that. Is commanding no torque a
possibility?

GuySmily, 2016-02-22:
Possible rules violation, as the slight power used trying to center the motor between poles might break the requirement of "shutting down the power to the motor."

I guess the questions are
1. Why does the motor controller still try to move the motor around at 0 torque?
2. Is that okay anyway?  Is _almost_ no power close enough to "shutting down power?"

alexsternberg, 2016-02-22:
Keep in mind that the motor will behave differently in the car. There may
not be sufficient power to move the inertia of the car but more than enough
for just the rotor.

GuySmily, 2016-02-22:
I believe there wouldn't be enough power to spin a pinwheel. However, that
still technically may not be rules compliant.

(Never replied from email before.. Let's see if this works)

GuySmily, 2016-02-27:
After discussing this in person, will not be disabling inverter.

closed 2016-02-27 by GuySmily