spartanracingelectric/VCU #57: Limit regen @ 300V

GuySmily, 2016-03-01:
Possibility of regen creating momentary voltage surge above 300V?

EV2.2.4 A violation is defined as using more than 80kW or exceeding the specified voltage for more than 100ms continuously or using more than 80kW or exceeding the specified voltage, after a moving average over 500ms is applied.

GuySmily, 2016-05-15:
Speed limit
Temperature
Voltage
amperage (32A)
Default (no pedal input) = 10% regen

GuySmily, 2016-05-15:
These limits should apply to both directions (regen and accel)

GuySmily, 2016-05-15:
Obey battery amperage burst limits
Ramp if possible

GuySmily, 2016-05-23:
2.2.3.2 CAN Torque command:
For a forward direction command:
 Positive torque command will give a positive torque feedback and is motoring for positive speed.
 Negative torque command will give a negative torque feedback and is regen for positive speed.
 Positive torque command will give a positive torque feedback and is regen for negative speed.
 A negative torque command should not be allowed if already going negative speed.

GuySmily, 2016-06-16:
All done except 300V limit (does mcm do this?)

chucktesta154k, 2016-06-16:
No. We just stop charging at 300

GuySmily, 2016-06-16:
Does the motor controller automatically limit it to 300v? Was this limit
programmed into eeprom already? Do we know/Are we just assuming that it
won't surge back up that high? If we're all safe then we can cut that
requirement out and label this as ready to test

chucktesta154k, 2016-06-16:
We could have an issue with regen while fully charged. So maybe its worth
trying to implemebt

closed 2023-09-22 by shibcreate