spartanracingelectric/VCU #126: TTC60-Motor Controller Relay Control Issue

vsawce, 2021-11-30:
`me->HVILOverride = (IO_RTC_GetTimeUS(me->timeStamp_HVILOverrideCommandReceived) < 1000000);` in motorController.c is driven TRUE after ~30 seconds (for 1 second, and reverts back to FALSE) post-startup on the TTC 60. This behavior causes the VCU to drive the motor controller's relay on and then off which isn't desired. The relay should be off during idle.

There doesn't seem to be a ghost 0x5FF debug message being displayed on the CAN bus, so it's likely related to the incorrect RTC usage or RTC library bug.

vsawce, 2022-01-21:
Temporarily disabled HVILOverride conditional statement. TTC60 basic functionality still operational without this feature, but might be worth looking into further debugging the HVILOverride

shibcreate, 2023-09-22:
Test HVILOverride for future bypass into HV for motorcontroller settings